Northeastern-Electric-Racing/Odysseus #10: Check Unix Time

evanlombardo, 2023-09-13:
Each message sent to Siren should contain a unix time field, which should have been sent within 5 minutes of the server receiving the message. If the time in the message is not within 5 minutes of the current epoch time on the server, the message should be ignored.

Example Message:
```
{
   "unix_time": 4362781,
   "topic": "MPU",
   "data": [
      {
         "name": "board_temperature",
         "val": 50,
         "units": "degC"
      },
      {
         "name": "system_state",
         "val": "OFF",
         "units": "enum"
      },
   ...
   ]
}
```

Harshal662, 2023-09-13:
From where it is sending/ generating the message?

evanlombardo, 2023-09-13:
> From where it is sending/ generating the message?

These messages will be generated when a client wants to subscribe or unsubscribe to a topic, or if a message is being published to a topic.

Also I'm not sure if these are really using epoch time but I'll confirm that in the morning.

jr1221, 2023-10-05:
Siren (websockets) abandoned

closed 2023-10-05 by jr1221