Northeastern-Electric-Racing/Odysseus #54: [Compilation] - Move CCACHE to directory within Repo

nwdepatie, 2024-01-13:
### Current Features

Currently we have a lot of cacheing directories used to speed up the build. However, these point to a directory tree outside of the repo itself.

### Desired Additional Features

I think we should move these caches to be within the local repo directory itself, and then add them into the gitignore. This will consolidate the buildroot build artifacts to be completely consolidated within the repo for easier removal and less setup on the users part to worry about

jr1221, 2024-01-23:
#62 takes care of this by the way of a docker volume.  Do u still think we should work this out for those building the old way?

jr1221, 2024-02-01:
docker supplants this

closed 2024-02-01 by jr1221