Northeastern-Electric-Racing/Odysseus #56: NanoMQ (NNG) segmentation faults when Calypso conducts initial payload send  (aarch64 buildroot nanomq only).

jr1221, 2024-01-13:
### Observed Behavior

Noted behavior:
Nanomq segfaults (with debug on/off, quic on/off) when connected to calypso in target environment.
Nanomq is fine with a host build when connected to host or target calypso.
Nanomq does not segfault with MQTTX test runs, only calypso generated messages.

**Cannot reporduce at all on x86_64.  Perhaps buildroot building for x86_64 could lead to the same issue?**

**Source failure: https://github.com/nanomq/NanoNNG/blob/56c3046bb1936d80cca0197b708c6f126cfa7e21/src/sp/protocol/mqtt/mqtt_parser.c#L165**

Line (approx, debugging symbols not working for some reason); ?? Need debug symbols!!!

### Expected Behavior

No segmentation fault. 

### Steps to Reproduce

1.  Start nanomq on buildroot aarch64 (any conf), mqtt://XXXX:1883
2.  Run calypso on any environment, pointing to the above URL.
3. Note segmentation fault and complete halt of all nanomq upon calypso init

**TODO tests**:

- Get debug symbols working and address santizer (valgrind) for more info if should be sent to NanoNNG 
- Attempt to reproduce on aarch64/pi OS
- Attempt to reproduce on x86_64 buildroot host infrastructure

### Screenshots (as needed)

Bug info:

```
#0  0x0000005592090178 in get_var_integer ()
No symbol table info available.
#1  0x00000055920905b0 in fixed_header_adaptor ()
No symbol table info available.
#2  0x000000559215050c in tcptran_pipe_recv_cb ()
No symbol table info available.
#3  0x0000005592086a30 in nni_taskq_thread ()
No symbol table info available.
#4  0x00000055920870f8 in nni_thr_wrap ()
No symbol table info available.
#5  0x00000055920893d0 in nni_plat_thr_main ()
No symbol table info available.
#6  0x0000007fbd970ce4 in ?? ()
No symbol table info available.
#7  0x0000007fb905f1a0 in ?? ()
No symbol table info available.
```

```
(gdb) info symbol 0x0000005592090178
get_var_integer + 16 in section .text of /home/jack/Projects/NER/buildroot/core/nanomq-5
```
```
(gdb) disas
Dump of assembler code for function get_var_integer:
   0x0000005592090168 <+0>:     ldr     w3, [x1]
   0x000000559209016c <+4>:     mov     x5, x0
   0x0000005592090170 <+8>:     mov     w4, #0x0                        // #0
   0x0000005592090174 <+12>:    mov     w0, #0x0                        // #0
=> 0x0000005592090178 <+16>:    ldrb    w7, [x5, w3, uxtw]
   0x000000559209017c <+20>:    mov     w6, #0x0                        // #0
   0x0000005592090180 <+24>:    mov     x2, #0x1                        // #1
   0x0000005592090184 <+28>:    and     w8, w7, #0x7f
   0x0000005592090188 <+32>:    add     w6, w6, #0x1
   0x000000559209018c <+36>:    lsl     x2, x2, #7
   0x0000005592090190 <+40>:    cmp     w6, w4
[... 166 lines cut ...]
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:1603 decode_pub_message: payload: [{"username":"(null)", "ts":1705176295451,"proto_name":"MQTT","keepalive":20,"return_code":"0","proto_ver":4,"client_id":"calypso","clean_start":0, "IPv4":"127.0.0.1"}], len = 166
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:1186 handle_pub: pipe_info size: [0]
2024-01-13 20:04:55 [3361] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:1176 nano_pipe_recv_cb: end of nano_pipe_recv_cb 0x5576d45e70
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/apps/broker.c:526 server_cb: WAIT ^^^^ ctx1 ^^^^
2024-01-13 20:04:55 [3361] TRACE ../nanomq/apps/broker.c:536 server_cb: total pipes: 0
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:1282 free_pub_packet: free topic
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:1298 free_pub_packet: free payload
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:1304 free_pub_packet: free pub_packet
2024-01-13 20:04:55 [3361] DEBUG ../nanomq/pub_handler.c:84 init_pipe_content: pub_handler: init pipe_info
2024-01-13 20:04:55 [3366] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:516 nmq_tcptran_pipe_send_cb:  ################ nmq_tcptran_pipe_send_cb [0x7f50000b70] ################ 
2024-01-13 20:04:55 [3365] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:606 tcptran_pipe_recv_cb: tcptran_pipe_recv_cb 0x7f50000b70

2024-01-13 20:04:55 [3361] TRACE ../nng/src/nng.c:390 nng_ctx_recv:  ######## nng_ctx_recv context id 1 ######## 
2024-01-13 20:04:55 [3366] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:530 nmq_tcptran_pipe_send_cb: tcp socket sent 4 bytes iov 0
2024-01-13 20:04:55 [3361] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:959 nano_ctx_recv: nano_ctx_recv start 0x5576d45e70
2024-01-13 20:04:55 [3366] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:1425 tcptran_pipe_send_start: ########### tcptran_pipe_send_start ###########
2024-01-13 20:04:55 [3366] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:1435 tcptran_pipe_send_start: aio not functioning
2024-01-13 20:04:55 [3366] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:902 nano_pipe_send_cb: ******** nano_pipe_send_cb 1257963498 ****
2024-01-13 20:04:55 [3365] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:631 tcptran_pipe_recv_cb: newly recevied 2 totoal received: 2  pos: 2 len : 0
2024-01-13 20:04:55 [3365] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:633 tcptran_pipe_recv_cb: still need byte count:0 > 0

2024-01-13 20:04:55 [3365] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:678 tcptran_pipe_recv_cb: pipe 0x7f50000b70 header got: e0 0 0 4 4d, 2!!

Segmentation fault
```

jr1221, 2024-01-13:
x86_64 log for same call:
```
2024-01-13 18:26:37 [235880] ERROR /build/source/nng/src/supplemental/nanolib/conf_ver2.c:1201 conf_parse_ver2: Configure file [(null)] or [/etc/nanomq.conf] not found or unreadable
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1141 print_conf: This NanoMQ instance configured as:
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1141 print_conf: This NanoMQ instance configured as:
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1144 print_conf: tcp url:                  nmq-tcp://0.0.0.0:1883 
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1144 print_conf: tcp url:                  nmq-tcp://0.0.0.0:1883 
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1147 print_conf: websocket url:            nmq-ws://0.0.0.0:8083/mqtt
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1147 print_conf: websocket url:            nmq-ws://0.0.0.0:8083/mqtt
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1165 print_conf: daemon:                   false
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1165 print_conf: daemon:                   false
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1167 print_conf: num_taskq_thread:         24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1167 print_conf: num_taskq_thread:         24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1169 print_conf: max_taskq_thread:         24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1169 print_conf: max_taskq_thread:         24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1171 print_conf: parallel:                 24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1171 print_conf: parallel:                 24
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1172 print_conf: property_size:            32
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1172 print_conf: property_size:            32
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1173 print_conf: max_packet_size:          10485760
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1173 print_conf: max_packet_size:          10485760
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1174 print_conf: client_max_packet_size:   10485760
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1174 print_conf: client_max_packet_size:   10485760
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1176 print_conf: msq_len:                  2048
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1176 print_conf: msq_len:                  2048
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1177 print_conf: qos_duration:             10
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1177 print_conf: qos_duration:             10
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1178 print_conf: keepalive_backoff:        1.500000
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1178 print_conf: keepalive_backoff:        1.500000
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1214 print_conf: allow_anonymous:          true
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1214 print_conf: allow_anonymous:          true
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1218 print_conf: acl_nomatch:              allow
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1218 print_conf: acl_nomatch:              allow
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1220 print_conf: enable_acl_cache:         on
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1220 print_conf: enable_acl_cache:         on
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1222 print_conf: acl_cache_max_size:       32
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1222 print_conf: acl_cache_max_size:       32
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1224 print_conf: acl_cache_ttl:            60
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1224 print_conf: acl_cache_ttl:            60
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1225 print_conf: acl_deny_action:          ignore
2024-01-13 18:26:37 [235880] INFO  /build/source/nng/src/supplemental/nanolib/conf.c:1225 print_conf: acl_deny_action:          ignore
2024-01-13 18:26:37 [235880] INFO  /build/source/nanomq/apps/broker.c:1281 store_pid: 235880
[... cut ...]
2024-01-13 18:26:39 [235890] TRACE /build/source/nng/src/sp/transport/mqtt/broker_tcp.c:860 tcptran_pipe_recv_cb: end of tcptran_pipe_recv_cb: synch! 0x7ff058000b70

2024-01-13 18:26:39 [235890] DEBUG /build/source/nanomq/apps/broker.c:490 server_cb: WAIT ^^^^ ctx2 ^^^^
2024-01-13 18:26:39 [235890] DEBUG /build/source/nanomq/apps/broker.c:557 server_cb: broker has nothing to do
2024-01-13 18:26:39 [235890] TRACE /build/source/nng/src/nng.c:372 nng_ctx_recv:  ######## nng_ctx_recv context id 2 ######## 
2024-01-13 18:26:39 [235890] TRACE /build/source/nng/src/sp/protocol/mqtt/nmq_mqtt.c:929 nano_ctx_recv: nano_ctx_recv start 0x1708ac0

```

jr1221, 2024-01-14:
Seg fault debug steps:
Turn off optimization (-g0), enable debug all, enable full gdb package, enable valgrind.
nanomq: enable debug, ptrace,  use -dcmake_build_type=debug, BR2_TOOLCHAIN_EXTRA_LIBS="libasan liblsan libtsan libubsan", strip exclude files "nanomq"

`gdb nanomq`
`set follow-fork-mode child`
on other terminal `gdb attach` to other child, `c`, then get stack and core dump in same dir.

jr1221, 2024-01-14:
Doesnt fail with:
```
BR2_TOOLCHAIN_EXTRA_LIBS="libasan liblsan libtsan libubsan"
BR2_ENABLE_DEBUG=y
BR2_STRIP_EXCLUDE_FILES="nanomq"
BR2_OPTIMIZE_0=y
BR2_PACKAGE_GDB=y
BR2_PACKAGE_GDB_DEBUGGER=y
BR2_PACKAGE_VALGRIND=y
BR2_PACKAGE_NANOMQ_DEBUG=y
BR2_PACKAGE_NANOMQ_TRACE=y
```
Must clean build to determine more failures.

mqtt_parser.c:165

Vars:
```
(gdb) info args
buf = 0x7f58000ba3 "7"
pos = 0x7f87ace7cc
```
```
(gdb) info locals
temp = 88 'X'
result = 127
p = 1476400224
i = 127
```
179
```
temp = 55 '7'
result = 55
p = 1
i = 0
```

jr1221, 2024-01-14:
Still works with BR2_OPTIMIZE_0=y.

jr1221, 2024-01-14:
Working -o0.
```
# nanomq start --log_level=trace
2024-01-14 19:18:00 [710] TRACE ../nng/src/core/taskq.c:289 nni_taskq_sys_init: result is 8
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1144 print_conf: This NanoMQ instance configured as:
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1147 print_conf: tcp url:                  nmq-tcp://0.0.0.0:1883 
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1150 print_conf: websocket url:            nmq-ws://0.0.0.0:8083/mqtt
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1168 print_conf: daemon:                   false
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1170 print_conf: num_taskq_thread:         8
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1172 print_conf: max_taskq_thread:         8
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1174 print_conf: parallel:                 8
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1175 print_conf: property_size:            32
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1176 print_conf: max_packet_size:          10240
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1177 print_conf: client_max_packet_size:   10240
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1179 print_conf: max_mqueue_len:           2048
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1180 print_conf: max_inflight_window:      2048
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1181 print_conf: max_awaiting_rel:         10s
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1182 print_conf: await_rel_timeout:        10s
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1183 print_conf: retry_interval:           10s
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1184 print_conf: keepalive_multiplier:     1250.000000
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1188 print_conf: http server port:         8081
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1189 print_conf: http server limit_conn:   2
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1190 print_conf: http server username:     admin
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1192 print_conf: http server auth type:    basic
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1200 print_conf:         public key file:      /etc/certs/jwt/jwtRS256.key.pub
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1220 print_conf: allow_anonymous:          true
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1224 print_conf: acl_nomatch:              allow
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1226 print_conf: enable_acl_cache:         on
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1228 print_conf: acl_cache_max_size:       32
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1230 print_conf: acl_cache_ttl:            60
2024-01-14 19:18:00 [710] INFO  ../nng/src/supplemental/nanolib/conf.c:1231 print_conf: acl_deny_action:          ignore
2024-01-14 19:18:00 [710] INFO  ../nanomq/apps/broker.c:1340 store_pid: 710
2024-01-14 19:18:00 [710] DEBUG ../nng/src/supplemental/nanolib/mqtt_db.c:385 dbtree_node_new: New node: []
2024-01-14 19:18:00 [710] DEBUG ../nng/src/supplemental/nanolib/mqtt_db.c:385 dbtree_node_new: New node: []
2024-01-14 19:18:00 [710] DEBUG ../nanomq/apps/broker.c:912 broker: db init finished
2024-01-14 19:18:00 [710] DEBUG ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:1360 nng_nmq_tcp0_open: open up nmq tcp0 protocol.
2024-01-14 19:18:00 [710] DEBUG ../nng/src/sp/protocol.c:40 nni_proto_mqtt_open:  nng internal init finished! Listener start
2024-01-14 19:18:00 [710] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:325 nano_ctx_init: &&&&&&&& nano_ctx_init 0x5591902328 &&&&&&&&&
2024-01-14 19:18:00 [710] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:527 nano_sock_init: ************* nano_sock_init 0x5591902218 *************
2024-01-14 19:18:00 [710] DEBUG ../nng/src/sp/protocol.c:42 nni_proto_mqtt_open: mqtt socket opened!
2024-01-14 19:18:00 [710] DEBUG ../nanomq/apps/broker.c:921 broker: listener init finished
2024-01-14 19:18:00 [710] DEBUG ../nanomq/apps/broker.c:927 broker: HTTP service initialization
2024-01-14 19:18:00 [710] DEBUG ../nanomq/apps/broker.c:937 broker: HTTP init finished
2024-01-14 19:18:00 [710] DEBUG ../nanomq/apps/broker.c:943 broker: webhook init finished
2024-01-14 19:18:00 [710] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:325 nano_ctx_init: &&&&&&&& nano_ctx_init 0x5591903e70 &&&&&&&&&
2024-01-14 19:18:00 [710] DEBUG ../nanomq/pub_handler.c:84 init_pipe_content: pub_handler: init pipe_info
2024-01-14 19:18:00 [710] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:325 nano_ctx_init: &&&&&&&& nano_ctx_init 0x5591904250 &&&&&&&&&
2024-01-14 19:18:00 [710] DEBUG ../nanomq/pub_handler.c:84 init_pipe_content: pub_handler: init pipe_info
2024-01-14 19:18:00 [710] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:325 nano_ctx_init: &&&&&&&& nano_ctx_init 0x55919045c0 &&&&&&&&&
[... cut ...]
2024-01-14 19:18:03 [719] DEBUG ../nanomq/apps/broker.c:526 server_cb: WAIT ^^^^ ctx2 ^^^^
2024-01-14 19:18:03 [721] TRACE ../nng/src/sp/transport/mqtt/broker_tcp.c:867 tcptran_pipe_recv_cb: end of tcptran_pipe_recv_cb: synch! 0x7f80000b70

2024-01-14 19:18:03 [719] DEBUG ../nanomq/apps/broker.c:593 server_cb: broker has nothing to do
2024-01-14 19:18:03 [719] TRACE ../nng/src/nng.c:390 nng_ctx_recv:  ######## nng_ctx_recv context id 2 ######## 
2024-01-14 19:18:03 [719] TRACE ../nng/src/sp/protocol/mqtt/nmq_mqtt.c:959 nano_ctx_recv: nano_ctx_recv start 0x5591904250
```

jr1221, 2024-01-16:
Works with -O1, with -Os breaks.

jr1221, 2024-01-17:
How to debug:
```
BR2_ENABLE_DEBUG=y
BR2_ENABLE_RUNTIME_DEBUG=y
BR2_STRIP_EXCLUDE_FILES="nanomq"
BR2_PACKAGE_GDB=y
BR2_PACKAGE_GDB_DEBUGGER=y
BR2_PACKAGE_GDB_TUI=y
BR2_PACKAGE_VALGRIND=y
```
`gdb nanomq`
`run start`
`calypso mqtt localhost:1883 vcan0 skip_can_configure`
`scp   -r ./output/build/nanomq-9828d7b0c432d9495c5f940d75df0a621203b814/ root@10.42.0.119:/`
`(gdb) dir /root/nanomq-9828d7b0c432d9495c5f940d75df0a621203b814/buildroot-build`

JaylinYu, 2024-01-17:
Could you fire and issue with related information to nanoMQ repo too?

jr1221, 2024-01-17:
> Could you fire an issue with related information to nanoMQ repo too? Perhaps a hidden security issue, although seems like related to compiler optimization

Alright, I have [here](https://github.com/nanomq/nanomq/issues/1607)  is one open on nanoMQ and [here](https://github.com/nanomq/NanoNNG/issues/812) is one for nanoNNG I previously made.  Feel free to let me know on either of those if you want me to take any further debugging steps.

jr1221, 2024-01-25:
Fixed in nng in https://github.com/nanomq/NanoNNG/pull/826.

Will release calypso pr #43 with this fix once it hits main, as well as plugin support.

closed 2024-01-25 by jr1221