Northeastern-Electric-Racing/Odysseus #69: [Git] - NRC Submodule cloned twice

nwdepatie, 2024-01-24:
### Observed Behavior

When you clone the repo for the first time, it seems like the NRC repo is being cloned twice

### Expected Behavior

Repo should onyl be cloned twice

### Steps to Reproduce

1. Clone repo fresh
2. Submodule update --init

jr1221, 2024-01-24:
One is nrc7292 and another is nrc7394.  Although we don't have the hw for 7394 i figured ill add it just in case we needed to figure out build issue with it it will be ready.

```
Submodule 'odysseus/buildroot' (https://gitlab.com/buildroot.org/buildroot.git) registered for path 'odysseus/buildroot'
Submodule 'odysseus/odysseus_tree/sources/Nero-2.0' (https://github.com/Northeastern-Electric-Racing/Nero-2.0) registered for path 'odysseus/odysseus_tree/sources/Nero-2.0'
Submodule 'odysseus/odysseus_tree/sources/nrc7292_sw_pkg' (https://github.com/newracom/nrc7292_sw_pkg) registered for path 'odysseus/odysseus_tree/sources/nrc7292_sw_pkg'
Submodule 'odysseus/odysseus_tree/sources/nrc7394_sw_pkg' (https://github.com/newracom/nrc7394_sw_pkg) registered for path 'odysseus/odysseus_tree/sources/nrc7394_sw_pkg'
Cloning into '/home/jack/Projects/NER/Siren/odysseus/buildroot'...
Cloning into '/home/jack/Projects/NER/Siren/odysseus/odysseus_tree/sources/Nero-2.0'...
Cloning into '/home/jack/Projects/NER/Siren/odysseus/odysseus_tree/sources/nrc7292_sw_pkg'...
Cloning into '/home/jack/Projects/NER/Siren/odysseus/odysseus_tree/sources/nrc7394_sw_pkg'...
Submodule path 'odysseus/buildroot': checked out '96d0d677790e659b822761b50561b0316b4abf43'
Submodule path 'odysseus/odysseus_tree/sources/Nero-2.0': checked out '5f89e3d71b83a44ea2de58702fe68eb8fc136bcb'
Submodule path 'odysseus/odysseus_tree/sources/nrc7292_sw_pkg': checked out '1dd6dfb3fb3b552bbc9b47529d60123a141ba0cb'
Submodule path 'odysseus/odysseus_tree/sources/nrc7394_sw_pkg': checked out 'f02ed64bf2ab42567813ec2ed8916f5b0c90dabd'
```

nwdepatie, 2024-01-25:
Oopsies, I'll close this

closed 2024-01-25 by nwdepatie