Northeastern-Electric-Racing/Odysseus #76: [Validation] - NanoMQ configuration

jr1221, 2024-02-01:
### Description

We need nanomq conf.

### Acceptance Criteria

It needs:

- bridging over quic
- tcp client connections
- sqlite persistence and log rotation (see #68 as alternative and/or supplant)
- no acl
- comments for performance tuning to come later
- plugin loading from `/usr/lib/nanomq/plugin_user_property_timestamp.so`

should be tested over halow link with random data.

### Proposed Solution

Add a conf file in the device tree overlay for the tpu (and nero?).

jr1221, 2024-02-01:
@evanlombardo any chance you may have some conf files I can work off of?  Ik u had trouble building it so..

evanlombardo, 2024-02-01:
> @evanlombardo any chance you may have some conf files I can work off of? Ik u had trouble building it so..

yes, check slack in a few minutes

closed 2024-02-14 by jr1221